ghidraninja/game-and-watch-backup #1: Ubuntu 20.04: Unable to locate package gdb-arm-none-eabi

ntrpnr, 2020-11-29:
```
sudo apt-get install gcc-arm-none-eabi binutils-arm-none-eabi gdb-arm-none-eabi openocd python3
[sudo] password for erik: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
**E: Unable to locate package gdb-arm-none-eabi**
```

I installed gdb-multiarch instead as described [here](https://askubuntu.com/a/1033056). Let's see if it works.

kbeckmann, 2020-11-29:
`gdb` is not required to run the scripts, however you probably want it at a later point when developing. For some reason it seems that this package has been removed. `gdb-multiarch` usually works pretty okay, however I have not tested it personally with this microcontroller.

This thread discusses the issue and proposes a workaround as well: https://askubuntu.com/questions/1243252/how-to-install-arm-none-eabi-gdb-on-ubuntu-20-04-lts-focal-fossa

nezza, 2020-11-30:
Thanks, I'll adjust the docs accordingly! We don't need the `gdb`, but I didn't test which package contains arm-none-eabi-objdump

closed 2020-11-30 by nezza

huanglilong, 2022-09-17:
http://ports.ubuntu.com/pool/universe/g/gcc-arm-none-eabi/gcc-arm-none-eabi_10.3-2021.07-4_arm64.deb

huanglilong, 2022-09-17:
http://ftp.de.debian.org/debian/pool/main/g/gcc-arm-none-eabi/gcc-arm-none-eabi_10.3-2021.07-4_amd64.deb