ghidraninja/game-and-watch-backup #3: Recover from error in Step 3 - Device won't turn on

ntrpnr, 2020-11-30:
After successfully running step 1 and step 2, step 3 failed:

[image]

**3_openocd.log**
```
Info : clock speed 1800 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.187426
Info : stm32h7x.cpu0: hardware has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32h7x.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x81000003 pc: 0xfffffffe msp: 0x200112b0
dumped 131072 bytes in 1.664896s (76.882 KiB/s)
```

**flashloader.log**
```
Info : clock speed 1800 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.187426
Info : stm32h7x.cpu0: hardware has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32h7x.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Resetting device
Programming ELF
19824 bytes written at address 0x20000000
downloaded 19824 bytes in 0.277210s (69.836 KiB/s)

target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0xfffffffe msp: 0xfffffffc
Loading image into RAM
1048576 bytes written at address 0x24000000
downloaded 1048576 bytes in 14.136097s (72.439 KiB/s)

sp (/32): 0x20020000

pc (/32): 0x20003559
```

At this stage, the device won't turn on. I've tried multiple power cycles for both device and ST-Link.
I've also kept the power button pressed for several minutes.

Any attempt to run step 3 again results in `Error: init mode failed (unable to connect to the target)`

**Is there any way to recover?**

nezza, 2020-11-30:
Hey, so a couple of things:

This commonly happens when you don't fully power cycle the device (ensure that really all power is gone from the Game and Watch), and/or if the power button is not pressed, so try that

If that does not work you can recover the original flash by running

./scripts/flashloader.sh stlink backups/flash_backup.bin (along those lines, on mobile will check later)

Cheers!

nezza, 2020-11-30:
Ah, the "init mode failed (unable to connect to target)" can indicate that you need to press & hold the power button while running the whole script

ntrpnr, 2020-11-30:
I followed your recovery instructions for step 3. It worked!

Let's close this one. Great work @nezza

kbeckmann, 2020-11-30:
Great that you got it working!

closed 2020-11-30 by kbeckmann

nezza, 2020-11-30:
Awesome, thanks for the feedback and glad it's working!