ghidraninja/game-and-watch-backup #7: Mac os st link issue on step 2

KsAmJ, 2020-11-30:
I followed the guide and done the exact connections and I am using an st link... [image]

BUT step 2 always fail... Showing error.. Any ideas or is there some packages or drivers needed for the st link on the mac

BTW I'm on macos big sur and step one run successfully

kbeckmann, 2020-11-30:
Can you post your logs? They are stored in the `logs` directory.

A common problem is that the openocd version in ubuntu is too old and doesn't have support for this mcu.

KsAmJ, 2020-12-01:
`Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
openocd/interface_stlink.cfg:1: Error: Can't find interface/stlink.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60
at file "openocd/flash_stlink.cfg", line 3
at file "openocd/interface_stlink.cfg", line 1
`

KsAmJ, 2020-12-01:
> Can you post your logs? They are stored in the `logs` directory.
> 
> A common problem is that the openocd version in ubuntu is too old and doesn't have support for this mcu.

`Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
openocd/interface_stlink.cfg:1: Error: Can't find interface/stlink.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60
at file "openocd/flash_stlink.cfg", line 3
at file "openocd/interface_stlink.cfg", line 1
`

SteveClement, 2020-12-01:
@KsAmJ this is probably due to openocd missing the config file for stlink.

I has the same issue on Debian and had to grab the latest package to make it work.

KsAmJ, 2020-12-01:
How did u manage to do that
Can u walk me through it?

kbeckmann, 2020-12-01:
> Error: Can't find interface/stlink.cfg in procedure 'script' at file "embedded:startup.tcl"

This indicates that you’re using an “old” version of openocd. You need to build it from git to get the new and good stuff.

nezza, 2020-12-01:
Yup, looks like OpenOCD is too old. Try `brew install --HEAD openocd` or so

closed 2020-12-01 by nezza