ghidraninja/game-and-watch-backup #9: Unable to restore after installing pwnadventure

kilrah, 2020-12-01:
After installing and playing a bit I wanted to go back to stock, but whatever I do the programmer (ST-Link V2) won't connect anymore, tried both with stlink firmware and with the jlink firmware, always "unable to reset target" (although it actually resets).

nezza, 2020-12-01:
Sounds like your electrical connection might be wonky?

kilrah, 2020-12-01:
Nope, really thoroughly checked, even used 2 different ST-Links... there was no issue for the backup, initial restore and pwnadventure flashing.
The ST-Link correctly identifies the chip, the chip resets when the stlink does a reset but it can't connect. Tried different speeds, and both openocd on linux as per the script and the st-link tool on windows with various settings for the reset type, low power debugging etc. The code might be doing something that prevents the debugger from successfully connecting...

nezza, 2020-12-01:
Are you holding down the power button while running the restore script?

kilrah, 2020-12-01:
Tried both with and without

nezza, 2020-12-01:
Can you try without the reset line connected?

kilrah, 2020-12-01:
Tried that already too :( Reset connected, unconnected and grounded

nezza, 2020-12-01:
Feel free to join our Discord to discuss: https://discord.gg/YM2crykp
The instructions worked for multiple people, and if the device is turned on it should be possible to connect via SWD, not sure what to do - in discord you can send pictures etc, maybe we spot something

KsAmJ, 2020-12-02:
hay i had exactly the same issue it is all about the timing of holding the power button and executing the script. it will work just cycle the power and try it. it took me 10 times then it worked eventually

kilrah, 2020-12-02:
Restoring the flash with a programmer helped in my case. 
But it could have been a wonky contact on the clips as well...

closed 2020-12-02 by kilrah

impeeza, 2021-12-28:
Hello, is a closed issue but without a really solution, what I did:

ran 
1_sanity_check.sh
2_backup_flash.sh
3_backup_internal_flash.sh
4_unlock_device.sh
5_restore.sh
all without any problem.  restart the device and test.

without discconect my Raspberry Pi ran install_pwnadventure.sh and the PWN Adventure start ok, then try to run 5_restore.sh to restore my original firmware, but get the error:

```
Ok, restoring original firmware! (We will not lock the device, so you won't have to repeat this procedure!)
Restoring SPI flash...
Restoring SPI flash failed. Check debug connection and try again.
```

and now neither of the scripts run.

what did you did to restore your firmware

kilrah, 2021-12-28:
Don't remember but given my last answer I likely reprogrammed the chip externally using my TL866II.