ghidraninja/game-and-watch-backup #17: Error in Step 3 - backup of the internal flash failed

cecio, 2020-12-11:
Hey,
First of all let me thank you for the great project. It is really awesome!

I'd like to explain an issue I'm experiencing. The step 3 is keep failing with this message:

`The backup of the internal flash failed. Please try again.`

This is the openocd log:
```
Open On-Chip Debugger 0.10.0+dev-01514-ga8edbd020-dirty (2020-12-02-00:18)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : J-Link V11 compiled Nov 12 2020 10:07:08
Info : Hardware version: 11.00
Info : VTarget = 4.467 V
Info : clock speed 1800 kHz
Info : SWD DPIDR 0x6ba02477
Info : stm32h7x.cpu0: hardware has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32h7x.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
dumped 131072 bytes in 1.149703s (111.333 KiB/s)
```
I don't see errors in the log. Also, the dumped file has the correct size (131072) and it seems it contains some data (after an initial set of 0s).
Looking at the code it looks like the hash of the dumped file is not equal to the one expected by the package, and that's what is causing the error.
I'm just guessing, but may be that the version of the flash are in some way different (new version? Different region?) so that the hash is not matching at all?

From your point of view, do I have any way to validate (or refute) this hypothesis?
Thanks so much!

cecio, 2020-12-11:
mmmmmmm may be not...

I see that you are actually verifying the hash of the dumped flash before the "rebuild"...so it should be fine.

I tried to compare the content of the memory (directly from jlink) with the dumped image, and it looks correct. So the problem should be in the flashing of the "new_flash_image"...but from the log it looks fine (here I tried another version of openocd):

```
Open On-Chip Debugger 0.11.0-rc1+dev-00001-g0dd3b7fa6-dirty (2020-12-10-23:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : J-Link V11 compiled Nov 12 2020 10:07:08
Info : Hardware version: 11.00
Info : VTarget = 1.922 V
Info : clock speed 1800 kHz
Info : SWD DPIDR 0x6ba02477
Info : stm32h7x.cpu0: hardware has 8 breakpoints, 4 watchpoints
Error: stm32h7x.cpu0 -- clearing lockup after double fault
Polling target stm32h7x.cpu0 failed, trying to reexamine
Info : stm32h7x.cpu0: hardware has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32h7x.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Resetting device
Programming ELF
19824 bytes written at address 0x20000000
downloaded 19824 bytes in 0.171006s (113.209 KiB/s)

target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0xfffffffe msp: 0xfffffffc
Loading image into RAM
1048576 bytes written at address 0x24000000
downloaded 1048576 bytes in 9.053162s (113.110 KiB/s)

sp (/32): 0x20020000

pc (/32): 0x20003559

Starting flash process
```
I'm a bit puzzled....

kbeckmann, 2020-12-12:
> I'm just guessing, but may be that the version of the flash are in some way different (new version? Different region?) so that the hash is not matching at all?

So far, we haven't seen different versions of the internal flash.

Please join the discord for better support as people can help you in real-time.

Have you tried power cycling the device and pressing the button when running step 3? It may lock up get stuck in a bad state so that could help.

cecio, 2020-12-12:
Thanks!
Yes, I'll join discord.
Thanks again

closed 2020-12-12 by cecio