ghidraninja/game-and-watch-backup #19: Cheap debugger

Airmonkdev, 2020-12-29:
Hey guys, I'm new in reverse engineering but I'd like to hack my G&W so can you please point me to a working cheap debugger?

Thanks

MarcoDotIO, 2021-02-18:
Nothing specific, but if you’re really wanting to go cheap, something like a USB male ST-Link can be found with a simple search on eBay (If you need to stay under $20 USD). Though I recommend fetching the J-Link Edu, found on Adafruit, if you’re willing to spend around $70 USD on a debugger.

Airmonkdev, 2021-02-19:
Thanks for response!
I'm stuck because on the main page of this repo there is the "Supported Debuggers" section and I've read: "Programmers we had a lot of trouble with: J-Link EDU Mini (does not work), cheap ST-Link clones". :(

Also I have the new MacBook with M1 chip so not sure if all the packages has been ported on Homebrew...

I have a spare Raspy, maybe I can try connecting a ST-Link clone to the Raspy? 
I'd like to hack my G&W and run Tetris instead of Super Mario Bros. 2.... do you think is it possible? :)

MarcoDotIO, 2021-02-20:
> Thanks for response!
> I'm stuck because on the main page of this repo there is the "Supported Debuggers" section and I've read: "Programmers we had a lot of trouble with: J-Link EDU Mini (does not [...]

I have one too, and the repos needed with homebrew are 100% compatible with the M1 processor. And yeah, of course. There should be the Retro Go mod once you unlocked the GnW. Any NES game* will work. (* - Some games that need the weird controllers like the light gun of course don’t function with this type of emulation, but I assume you already knew that going in). And I’ve seen people using the knock off ST-Link, so it’s a hit or miss there, but I still recommend the J-Link Edu Regular because you can find it relatively cheap. Also of course test clips too.

Airmonkdev, 2021-02-21:
Thank you so much for all the info!
I gonna search for a J-Link Edu Regular as first and, since I'm newbie, hope someone of you could help me to put the Tetris rom instead of the SuperMario Bros 2 one. That's will be great for me! ;)

MarcoDotIO, 2021-02-22:
> Thank you so much for all the info!
> I gonna search for a J-Link Edu Regular as first and, since I'm newbie, hope someone of you could help me to put the Tetris rom instead of the SuperMario Bros 2 one. That's will be great for me! ;)
> 
> PS: this one will works?
> [https://www.amazon.it/ARCELI-Versione-Microcontroller-Emulatore-Programmatore/dp/B07RKVM7H8/ref=sr_1_2?__mk_it_IT=ÅMÅŽÕÑ&dchild=1&keywords=st+link&qid=1613926737&sr=8-2](https://www.amazon.it/ARCELI-Versione-Microcontroller-Emulatore-Programmatore/dp/B07RKVM7H8/ref=sr_1_2?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=st+link&qid=1613926737&sr=8-2)

I'd recommend this: https://www.adafruit.com/product/1369?gclid=CjwKCAiAyc2BBhAaEiwA44-wW4-6SfCNYCOJ4BFUHGRaytQzo_oGXfncRhtHf1vW_bB40gN446O0dxoCCa8QAvD_BwE

I would still say go on the discord provided on the Github page. And yeah all you really need are the dumped ROMs for the games: https://discord.gg/rE2nHVAKvn

Airmonkdev, 2021-02-23:
Thanks you so much! See you on Discord....

MarcoDotIO, 2021-03-01:
> Thanks you so much! See you on Discord....

No worries.

closed 2023-09-10 by kbeckmann