ghidraninja/game-and-watch-backup #22: Step 2 fails to validate flash backup checksum -- different checksum every time

arnemart, 2021-04-10:
First of all, thank you so much for putting this together.

When running step 2, the script always fails at the last step, because the checksum of flash_backup_checksummed.bin does not match. After trying the script several times (10+) and manually running shasum each time, I see that the checksum is different every time. The ITCM checksum is always fine.

- I'm using a j-link (probably a cheap clone) connected with test clips to SWDIO, SWDCLK and ground (tried connecting all 5 pins but same result)
- I've tried both macOS Big Sur with openocd 0.11.0 and Ubuntu with latest HEAD openocd built from source
- I've tried lowering connection speed to 100, replacing all cables, blowing in the cartridge slot

Anyone have any ideas what's wrong? I'm guessing my cheapo j-link might be the culprit but any other suggestions are welcome.

PolluxPT, 2022-06-03:
I'm getting this issue, how did you solved it?
I modded one before without this issue.
Thanks.

Einstein9781, 2024-02-18:
Push! Any resolution on this issue? Thanks.

nezza, 2024-02-18:
Most probably a connection issue, can't reproduce and all cases we've learned about ended up being a connection issue