ghidraninja/game-and-watch-backup #27: The backup of the internal flash failed

thierryfischer, 2021-12-02:
Hi,

At step 3 I got this error:
```
me@MYCOMPUTER:~$ sudo OPENOCD="/opt/openocd-git/bin/openocd" ./3_backup_internal_flash.sh stlink mario
This step will overwrite the contents of the SPI flash chip that we backed up in step 2.
It will be restored in step 5. Continue? (y/N)
yGenerating encrypted flash image from backed up data...
Programming payload to SPI flash...
Flash successfully programmed. Now do the following procedure:
- Disconnect power from the device
- Power it again
- Press the power button on the device
- The LCD should show a blue screen
- If it's not blue, you can try pressing the Time button on the device
- Press return

Dumping internal flash...
Verifying internal flash backup...
The backup of the internal flash failed. Please try again.
```
I did have blue screen once powered down & up and press enter after that.
The log file says:
```
Open On-Chip Debugger 0.11.0+dev-00515-g1ad6ed38b-dirty (2021-11-30-02:12)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
1300
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate

Info : clock speed 1800 kHz
Info : STLINK V2J17S4 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.243158
Info : stm32h7x.cpu0: Cortex-M7 r1p1 processor detected
Info : stm32h7x.cpu0: target has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32h7x.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0xfffffffe msp: 0xfffffffc
Info : Device: STM32H7Ax/7Bx
Info : assuming 2048k flash
Info : STM32H7 flash has dual banks
Info : Bank (0) size is 256 kb, base address is 0x08000000
Initializing Octo-SPI interface
Info : 4-byte address parameter table
Info : valid SFDP detected
Info : flash1 'sfdp' id = 0x3425c2 size = 1024kbytes
Error: FSIZE in DCR(1) doesn't match actual capacity.
** Programming Started **
** Programming Finished **
** Verify Started **
** Verified OK **
```

thierryfischer, 2021-12-02:
Solved it.
It came from the STLink V2 clone.
`Bus 001 Device 006: ID 0483:3748 STMicroelectronics ST-LINK/V2`
Using a Nucleo STLink did the trick:
`Bus 001 Device 007: ID 0483:374b STMicroelectronics ST-LINK/V2.1 (Nucleo-F103RB)`

timehlers, 2021-12-19:
I have the same problem (and I only have an STLink V2 clone:

Bus 002 Device 004: ID 0483:3748 STMicroelectronics ST-LINK/V2

Previously with Mario G&W every step went well. Now in the latest git version with Zelda G&W the resulting internal_flash_backup_zelda.bin from step 3 is always different, when I use:

acer:/home/tehlers/game-and-watch-backup # cat openocd/interface_stlink.cfg
source [find interface/stlink.cfg]
adapter speed 500
transport select hla_swd
source [find openocd/stm32h7x_spiflash.cfg]

reset_config none

When I use the old config (from the previous git version):

acer:/home/tehlers/game-and-watch-backup # cat openocd/interface_stlink.cfg
source [find interface/stlink.cfg]
transport select hla_swd
source [find target/stm32h7x.cfg]

reset_config none

At least the dump stays the same:

acer:/home/tehlers/game-and-watch-backup # shasum backups/internal_flash_backup_zelda.bin
484e7271b686c5f87412f6127ea07e90136eeb17  backups/internal_flash_backup_zelda.bin

But is not the required checksum.

Is there a way to use the STLink V2 clone on the Zelda version?

Thanks and best

Tim

toto99303, 2021-12-24:
+1
Solution was to use the original ST-Link nucleo.
The clone ST-Link v2 worked with Mario, but not with Zelda....

Raniita, 2023-01-25:
Hi! On my workaround with the ST-Link clone and Zelda unit, i face with this problem that is it unable to have a nice internal flash dump. To fix that, i upgrade the firmware of the ST-Link clone, after that, the dump goes nice!

Here the upgrader tool and some info about that: https://calinradoni.github.io/pages/200616-st-link-firmware-upgrade.html

Hope that some one could find useful