ghidraninja/game-and-watch-backup #29: Restore with new 64 mb chip failed

Ch1n01, 2022-08-06:
Hello!
I had 2 consoles, I backed up first console, hacked, changed the chip to 64 mb, restored the backup, recorded a bunch of games, everything is fine with this. Did all this in Ubuntu VMware under Mac OS. I also made a backup of the second console, hacked it, left the soldering of the chip for later, made a snapshot in VMware. Later I soldered the chip, and when I started to restore the firmware, I realized that I accidentally deleted the snapshot with the backup. So I decided to soldering 1mb chip back, made a backup, desoldering it and soldering 64mb chip again. After that when I trying to restore firmware on 64mb chip i am seeing this:

Restoring SPI flash...
./5_restore.sh: line 34: 60868 Aborted                 (core dumped) ${OPENOCD} -f "openocd/target_${TARGET}.cfg" -f "openocd/interface_${ADAPTER}.cfg" -c "init;" -c "halt;" -c "program backups/flash_backup_${TARGET}.bin 0x90000000 verify;" -c "exit;" >> logs/5_openocd.log 2>&1

In openocd log:

Open On-Chip Debugger 0.11.0+dev-00715-g480d4e177-dirty (2022-06-14-02:49)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
1300
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate

Info : clock speed 1800 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.290505
Info : [stm32h7x.cpu0] Cortex-M7 r1p1 processor detected
Info : [stm32h7x.cpu0] target has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32h7x.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080164b8 msp: 0x2001b5c0
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0801ad48 msp: 0x2001b620
Info : Device: STM32H7Ax/7Bx
Info : flash size probed value 128k
Info : STM32H7 flash has dual banks
Info : Bank (0) size is 256 kb, base address is 0x08000000
Initializing Octo-SPI interface
Error: timeout
Error executing event reset-end on target stm32h7x.cpu0:
embedded:startup.tcl:876: Error: 
in procedure 'program' 
in procedure 'ocd_process_reset' 
in procedure 'ocd_process_reset_inner' called at file "embedded:startup.tcl", line 876
** Programming Started **
Error: timeout
openocd: src/flash/nor/core.c:857: flash_write_unlock_verify: Assertion `run_size > 0' failed.

Can I something to do with this? Thanks!

BenjaminSoelberg, 2023-08-21:
I have created a pull request to fix this

BenjaminSoelberg, 2023-08-22:
I know this is a year old by now but could you try with the latest version ?
Just prefix your restore command with LARGE_FLASH=1 and you should be good.

Ch1n01, 2023-08-22:
Hello Benjamin! Yes, of course, I can check as soon as I get back from vacation in a couple of weeks. Thanks for the answer!

BenjaminSoelberg, 2023-10-09:
Did it work ?